tonuino/TonUINO-TNG #16: Shortcut at startup doesn't work

dl9sec, 2022-11-07:
Hi,
I used the latest sourcecode (TonUINO Classic, 5 buttons) and selected a file (folder 1, title 1) to be played once at startup in the admin mode at shortcuts.
Box told me, that it is stored, but the title isn't played at startup of the box, just a "pling" sound is played...

Regards, Thorsten

boerge1, 2022-11-07:
Created branch 'issue_016' with a fix. @dl9sec could you please test it?
Created Pull Request #17

dl9sec, 2022-11-07:
Thanks for the fast response.
Will give you a feedback tomorrow (TonUINO currently in use  by my son to fall asleep :-D )...
Regards, Thorsten

dl9sec, 2022-11-08:
The "issue_016"-branch seems to fix the problem.
Thank you.
Regards, Thorsten

closed 2022-11-09 by boerge1