tonuino/TonUINO-TNG #24: Repeat LongPress Action of a Button after next longPressPeriod

boerge1, 2022-11-25:
If I press a button long and do not release it, the action should repeated after the next longPress timeout. 
(as PC keyboard works)

seisfeld, 2022-11-25:
Why? IMHO TonUINO is a music for small childs. They most likely can't control how long they press. This leads to unwanted skipping through tracks (depending on play mode) as well as high volumes if they rest their hand on the volume key. Makes no sense IMHO.

boerge1, 2022-11-25:
It was a request for the case that you don't have switched the volume keys. Then you have to change the volume by long pressing the key again and again. And because of you have no response after long press is active this is a little bit annoying. With this feature you can say: press the key as long until the volume is what you want.

seisfeld, 2022-11-25:
Well so the root cause here is 3 buttons vs. 5 buttons really. If we only support 5 buttons this would mitigate all these edge cases and remove the requirement of switching functions and log presses during playback entirely. So we (well @xfjx really) need to decide on a common control scheme going forward.

boerge1, 2022-11-25:
Oh, I think we have many boxes in the field, that have only 3 Buttons. We should not finish supporting them. 
I already made a test version on Branch 'issue_024'. Please try. But of course, it is only interesting for the 3 Button case.

jochenberger, 2022-11-25:
> Why? IMHO TonUINO is a music for small childs. They most likely can't control how long they press.

Which age are do you mean here exactly? My 3-year-old can control that pretty well actually. 😉

> This leads to unwanted skipping through tracks (depending on play mode) as well as high volumes if they rest their hand on the volume key. 

The volume issues can be mitgated by setting a sensible maximum volume.

Maybe we could even offer to change the long press threshold? A actually reduced it to 400ms in my build.

jochenberger, 2022-11-25:
But do we really want that behavior for track skipping or shortcuts? I'm not really sure about that.

boerge1, 2022-11-27:
> But do we really want that behavior for track skipping or shortcuts? I'm not really sure about that.

The solution on branch issue_024 was only study how to solve it with JC_Button library. In a final solution the obove would be solved.

boerge1, 2022-11-29:
Will be continued on issue #28 and closed here

closed 2022-11-29 by boerge1