tonuino/TonUINO-TNG #25: DFMiniMp3 lib version Arduino

AlmightyFrog, 2022-11-27:
Hi,

i just stumbled over TonUINO-TNG, nice to see there is some development going on again! :)

Anyhow, would you kindly check whether https://github.com/tonuino/TonUINO-TNG/blob/06ea5aa5758514455e2610944584704fd6f7e37e/Tonuino.ino#L22 is still necessary?
Was it a technical restriction or only cause lib was not yet in arduino until mid of this year?

This year when I intended to fix Tonuino for me for bad Chip version, i asked @makuna here https://github.com/Makuna/DFMiniMp3/issues/94 to kindly add 1.1.0 to arduino lib which you can find here: https://www.arduinolibraries.info/libraries/df-player-mini-mp3-by-makuna.

Best Regards, AlmightyFrog

PS: Btw, @makuna I did miss to thank you in time and then felt bad to reopen your ticket: i thank you very very much for the fast action and it was really helpful for me back then to fast fix the player with a "bad" chip version! :heart:

boerge1, 2022-11-27:
Oh sorry. We already updated the software to support th new library version but forgott to update the comment in the tonuino.ino
Will remove the comment.

boerge1, 2022-11-27:
created pull request #27

closed 2022-12-07 by boerge1