tonuino/TonUINO-TNG #26: DFMiniMp3 lib support T_CHIP_VARIANT

AlmightyFrog, 2022-11-27:
As it seems you have already migrated to 1.1.0 version, how about support T_CHIP_VARIANT?

Here just a additional template parameter would be necessary, could be done by macro definition I guess.
https://github.com/tonuino/TonUINO-TNG/blob/5e10a49fbfc400ce63a240a97db27c9db041c976/src/mp3.hpp#L26 

If you chose "Mp3ChipOriginal", you still have the "old" behavior (with checksum), but switching to "Mp3ChipMH2024K16SS" enables to use also chip versions which have issues (no checksum).
(See: https://github.com/Makuna/DFMiniMp3/blob/9838e6d541d5c8b35e06aa4078adb49bc592e484/src/DFMiniMp3.h#L153 and https://github.com/Makuna/DFMiniMp3/blob/9838e6d541d5c8b35e06aa4078adb49bc592e484/src/DFMiniMp3.h#L175)

In my quick and dirty fix for a other tonuino for i just exactly did that by a macro in platformio.ini with -D setting parameter directly, but if we want to support still arduino we could also add a #ifndef->#def in general config and additional then make it possible to controll with platformio.ini.

boerge1, 2022-11-27:
Created pull request #27

AlmightyFrog, 2022-12-03:
Seems perfect, thanks!

I can't test it currently as I don't have a second player at hands currently, but code changes in PR #27 seems almost perfect.:
Only point i am not sure about is, whether you really want to make Mp3ChipMH2024K16SS as default or just did miss to comment the line 25 in [src/constants.hpp](https://github.com/tonuino/TonUINO-TNG/pull/27/files#diff-6dd9f08c662986128e0efaf41168bd97011c56c9554337eb5fd5151f6bfd1a18).
Or do i interpret it wrongly?

boerge1, 2022-12-04:
Oops, yes you are right. This line has to be commented out. I'm changing it.

closed 2022-12-07 by boerge1