tonuino/TonUINO-TNG #28: Revise Button behavior

boerge1, 2022-11-29:
Proposal: change the button behavior the same as the fork of [Thomas-Lehnert](https://discourse.voss.earth/t/tester-gesucht-tonuino-3-1/12086/62): 
```
three Button
============
                                                      vol/next-prev inverted
button     admin-menu  idle        pause       play        play
-------------------------------------------------------------------------
pause      select                  play        pause       pause
pauseLong  end adm     shortcut    shortcut    track#      track#
up         next                                next        vol++
upLong     next 10     shortcut    shortcut    vol++       next
down       prev                                prev        vol--
downLong   prev 10     shortcut    shortcut    vol--       prev
                       
five Button 
===========           
                                                      vol/next-prev inverted
button     admin-menu  idle        pause       play        play
-------------------------------------------------------------------------
pause      select                  play        pause       pause
pauseLong  end adm     shortcut    shortcut    track#      track#
up         next                                next        vol++
upLong     next 10     shortcut    shortcut    next 10     vol++ cont.
down       prev                                prev        vol--
downLong   prev 10     shortcut    shortcut    prev 10     vol-- cont.
four                                           vol++       next   
fourLong                                       vol++ cont. next 10
five                                           vol--       prev   
fiveLong                                       vol-- cont. prev 10
```
comments are welcome

boerge1, 2022-11-29:
Made a first implementation on branch 'issue_028'.

jochenberger, 2022-12-01:
I tried that but I wish we had
3-btn, Play upLong vol++ cont. and Play downLong vol-- cont. (see #24 and https://github.com/tonuino/TonUINO-TNG/commit/d75ada2a4f700a4a4dfb7a10f49c10bde573a22b#r90991535).

boerge1, 2022-12-07:
Created pull request #30

jochenberger, 2022-12-07:
> vol/next-prev inverted has no effect for 5 Buttons

Should the respective sections be removed from the admin menu then?

jochenberger, 2022-12-08:
> I tried that but I wish we had 3-btn, not inverted: Play upLong vol++ cont. and Play downLong vol-- cont. (see #24 and [d75ada2#r90991535](https://github.com/tonuino/TonUINO-TNG/commit/d75ada2a4f700a4a4dfb7a10f49c10bde573a22b#r90991535)).

Oh, you already did that, I didn't notice until I just tried to change the behavior. :laughing: Maybe you should say so in the comment.

boerge1, 2023-01-16:
Pull request #30 is merged

closed 2023-01-16 by boerge1