tonuino/TonUINO-TNG #36: Compilation error: 'SerialType' was not declared in this scope

r00tr4v3n, 2023-01-02:
Hello,
I tried compiling the latest master on Linux and Windows but I'm only getting 
`/home/user/Downloads/Tonuino-TNG/src/mp3.hpp:29:25: error: 'SerialType' was not declared in this scope`
Compile flag should be set to gnu++17.

```/home/user/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++17 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -std=gnu++17 -I/home/user/.arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino -I/home/user/.arduino15/packages/arduino/hardware/avr/1.8.6/variants/eightanaloginputs -I/home/user/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/EEPROM/src -I/home/user/Arduino/libraries/MFRC522/src -I/home/user/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src -I/home/user/Arduino/libraries/JC_Button/src /tmp/arduino-sketch-7DA08D924372CDCF19851B3CC89CFCDB/sketch/src/mp3.cpp -o /dev/null```

Adjusting the defines by hand does not yield any change. I feel like this might be an Arduino issue rather than the Tonuino code.

boerge1, 2023-01-02:
If you are using the Arduino IDE you have to edit the file constants.hpp and uncomment the define for your HW.

r00tr4v3n, 2023-01-02:
I tried that with classic and AIO. What IDE are you using?

boerge1, 2023-01-02:
Could you describe, what changes did you make in the file constants.hpp
I'm using platformio or eclipse.

r00tr4v3n, 2023-01-02:
Uncomment
`#define TonUINO_Classic`

boerge1, 2023-01-02:
Could you run the cmd-line in the very first comment from you. There should be long output. Could you send me the output? If there is not long output, could you change the option -o /dev/null to -o mp3.txt and send me this file.

r00tr4v3n, 2023-01-02:
There you go: [logpaste](https://www.logpasta.com/paste/raw/87504e56-dd0b-4e4f-a1c1-f556b5f1442a.txt)

boerge1, 2023-01-02:
In the file you can see, that you didn't uncomented the define:
```
# 5 "/home/user/Downloads/Tonuino-TNG/src/constants.hpp" 2

// Select the right PCB by uncommenting one of the following lines
// Bitte die passende Platine durch entfernen der Kommentare in einer der folgenden Zeilen auswählen
//#define TonUINO_Classic
//#define ALLinONE
//#define ALLinONE_Plus
```

r00tr4v3n, 2023-01-03:
Weird, both Arduino IDEs  (v1.8 and v2.0) do not store made changes. Using an editor worked, thank you.

closed 2023-01-03 by r00tr4v3n