tonuino/TonUINO-TNG #37: Compilation error

PieterJanSterk, 2023-01-02:
I get this error: 

```
C:\Users\pieter-jan.sterk\Desktop\tonUINO\TonUINO-TNG-main\Tonuino-TNG\src\chip_card.cpp:52:66: error: 'printStatusCode' function uses 'auto' type specifier without trailing return type
 auto printStatusCode(MFRC522& mfrc522, MFRC522::StatusCode status) {
                                                                  ^
C:\Users\pieter-jan.sterk\Desktop\tonUINO\TonUINO-TNG-main\Tonuino-TNG\src\chip_card.cpp:52:66: note: deduced return type only available with -std=c++14 or -std=gnu++14
C:\Users\pieter-jan.sterk\Desktop\tonUINO\TonUINO-TNG-main\Tonuino-TNG\src\chip_card.cpp:59:65: error: 'printPiccType' function uses 'auto' type specifier without trailing return type
 auto printPiccType(MFRC522& mfrc522, MFRC522::PICC_Type piccType) {
                                                                 ^
C:\Users\pieter-jan.sterk\Desktop\tonUINO\TonUINO-TNG-main\Tonuino-TNG\src\chip_card.cpp:59:65: note: deduced return type only available with -std=c++14 or -std=gnu++14
exit status 1
Fout bij het compileren voor board Arduino Nano
```
I edited constants.hpp, correctly I think, what am I doing wrong here?

boerge1, 2023-01-03:
Please read the readme file:

die Datei 'platform.local.txt' muss in den avr Ordner kopiert werden. Dieser Ordner ist gewöhnlich folgender:

    Windows: C:\Users<Benutzer>\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6

    Linux: /usr/share/arduino/hardware/archlinux-arduino/avr

PieterJanSterk, 2023-01-03:
I didn't have a avr or a 1.8.6 folder here, I created one. Is this wrong? Should this be present?

boerge1, 2023-01-03:
What HW you are compiling for?
What OS do you have?
Could you please search your disk for the file 'platform.txt' and reply here all the locations of this file?

PieterJanSterk, 2023-01-03:
I'm compiling for the Tonuino "Classic", running on an Arduino Nano. 
Running Windows 10. 
My AppData\Local\Arduino15\packages\arduino\hardware did not have an avr or a 1.8.6 folder, so I created these and copied the file. I also installed platform.io, is there a way to check if I installed this correctly? I havent installed the platform.io shell commands, perhaps this is the mistake? 

I ran a search on my C drive for the search parameter "platform.txt". It came back empty. 

I'm very sorry if this is a stupid error on my end, I hope I'm not wasting anyone's time on this.

dl9sec, 2023-01-04:
Hi Pieter,
you have to install the "Arduino AVR Boards" in the board manager.
Then put the attached file to the directory "C:\Users\<USER>\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6".

Regards, Thorsten

[platform.local.txt](https://github.com/tonuino/TonUINO-TNG/files/10342656/platform.local.txt)

PieterJanSterk, 2023-01-04:
This was my fault, stupid. Sorry to bother!

closed 2023-01-04 by PieterJanSterk