tonuino/TonUINO-TNG #43: Missing mp3 files 262.mp3 referenced in mp3.hpp and state_machine.cpp

tommywienert, 2023-01-14:
Hi there, when building the firmware on your own (without the webpage), the dfplayer reports error 6 for playing file 262.
This is happening because the files 262 of directory advert and mp3 are missing, but are in the code

mp3.hpp:
https://github.com/tonuino/TonUINO-TNG/blob/main/src/mp3.hpp#L34
https://github.com/tonuino/TonUINO-TNG/blob/main/src/mp3.hpp#L124

advert.cpp:
https://github.com/tonuino/TonUINO-TNG/blob/main/src/state_machine.cpp#L687
https://github.com/tonuino/TonUINO-TNG/blob/main/src/state_machine.cpp#L855

The file is missing in the sdcard.zip that is offered on the webpage
https://tonuino.de/sdcard_de.zip

commenting out those lines resolves the issue and cards will be played.

greetings
Tommy

boerge1, 2023-01-14:
You have to change the mp3 and advert directories vir version 3.1. See the Readme to find the url to download this.

tommywienert, 2023-01-14:
whoops, my bad.
Maybe could add that info to the webpage ;) thanks

closed 2023-01-14 by tommywienert

PieterJanSterk, 2023-01-14:
So, just to be clear, because I am currently translating all prompts to Dutch: which folder contains the files up to +/- 300, and which folder contains the files up to +/- 900?