tonuino/TonUINO-TNG #45: audio_messages_de.txt isn't up to date.

PieterJanSterk, 2023-01-15:
I was translating the audiofiles to Dutch, and noticed this file is not up to date. It's missing 0320, 0321, 0402, 0913 and 0919. The freeze dance in the advert folder, 0301, is described as 3 separate files, but is in reality one file?

PieterJanSterk, 2023-01-15:
Also, I'm very sorry, I'm no programmer and don't use Github al lot, I don't know there is a way I can add these changes? Would to help!

boerge1, 2023-01-15:
I already updated the audio_messages_de.txt in the branch [issue_007](https://github.com/tonuino/TonUINO-TNG/tree/issue_007). The pull request [14](https://github.com/tonuino/TonUINO-TNG/pull/14) is not ready to merge for another reason. 
So, could you take the file from branch issue_007? You can send me the result file via PM or you could make a pull request from a fork.

seisfeld, 2023-01-15:
#14 or issue_007 should not contain audio_messages_de.txt changes anyways IMHO. This should purely be to align the script with the upstream version. The messages itself should be treated as a separate issue, because currently the structure is outdated and it's entirely unclear how this will turn out in the future.

boerge1, 2023-01-15:
> #14 or issue_007 should not contain audio_messages_de.txt changes anyways IMHO. This should purely be to align the scripts with the upstream version. The messages itself should be treated as a separate issue (if at all at this point), because currently the structure is outdated and it's entirely unclear how this will turn out in the future.

Yes, you are right! I make a new pull request for the audio files only.

PieterJanSterk, 2023-01-15:
Sorry, this is all new to me, thanks!

boerge1, 2023-01-15:
Oh no, was my fault. 
This issue is now fixed with pull request #47. 
You can send me the translated file via PM or you can make a pull request from a fork.

closed 2023-01-18 by boerge1

PieterJanSterk, 2023-01-19:
I made a fork with the files! Again, not sure that everything is right, but at least the files are now public and open.

https://github.com/PieterJanSterk/TonUINO-TNG-Dutch-audio-files

PieterJanSterk, 2023-01-19:
If the "audio_messages_de.txt" is updated, I will try to add a "audio_messages_nl.txt"

boerge1, 2023-01-20:
The file "audio_messages_de.txt" is already updated in the main branch via pull request #47