tonuino/TonUINO-TNG #53: Party Modus always shuffles first 31 songs

PieterJanSterk, 2023-01-31:
The Party Modus only seems to shuffle the first 31 songs, whether there are more, or fewer songs.

I have a folder with 18 songs, everything above 18 gets a DF Player: Error 6, which makes perfect sense of course. Another folder, which contains 193 files, only shuffles and plays the first 31. 

I checked the correct labeling of everything, and there are no gaps in numbering. I tried writing the cards again, deleting the folders and writing the files again, ...  I'm on Windows, and never used the SD card on a Mac. So as far is I can tell, no extra hidden files in the folders, using Windows File Explorer. 

I checked the "old" firmware, this issue:

https://github.com/xfjx/TonUINO/issues/27

seems a bit like mine, but not quite. Am I the only one experiencing this, is this standard behavior, ...?

PieterJanSterk, 2023-02-01:
Okay, so far, things I tried:

- Replacing the Leiterkartenpiraten DF player with another Leiterkartenpiraten DF player: This did not change anything.
- Special party mode: A great workaround! This works swell for folders which are "misread" as MORE files. However, in folders with a lot of files, and a lower read number, this does not work. The special mode only lets me select the top read number, not the actual top number. 
- Copying the SD card and writing a new, fresh formated card: I try using branded SD cards, so far I've tried a Sandisk Ultra and Sandisk Extreme Pro.

None of my filenames on the entire card have titles beyond a number. So just 001, 002, 003, ... instead of 001 - Song 1, 002 - Song 2, .... They do however still have songtitle and artist as metadata. I have not tried deleting all this metadata, for obvious reasons.

Is this a bug in the way the the number of files is read? Something with the `numTracksInFolder` value? I read the bug fixes, but I'm not quite sure how to implement them for the 3.1 firmware. 

In the "tonuino.cpp" file, I found these lines: 

```
  case mode_t::party:
    // Party Modus: Ordner in zufälliger Reihenfolge
    myFolder->special = 1;
    myFolder->special2 = numTracksInFolder;
    __attribute__ ((fallthrough));
    /* no break */
  case mode_t::party_vb:
    // Spezialmodus Von-Bis: Party Ordner in zufälliger Reihenfolge
    LOG(play_log, s_info, F("Party"));
    LOG(play_log, s_info, myFolder->special, str_bis(), myFolder->special2);
    mp3.enqueueTrack(myFolder->folder, myFolder->special, myFolder->special2);
    mp3.shuffleQueue();
    break;
```

	
Can I change the `myFolder->special2 = numTracksInFolder` value here to a standard 255, and use the Special Party mode for all Party folder, or will this break other code along the way?

boerge1, 2023-02-02:
If you are using the dfplayer from Leiterkartenpiraten, then the player should not be the problem.
Did you already tried to rewrite the SD card or even better to try another SD card?
You do not have to change the code if using the special party mode. But maybe you have to write the chip card with an Android App. This is because you can select the second parameter (bis = special2) only up to the value the dfplayer gives as number of files in folder.

PieterJanSterk, 2023-02-04:
I really like the Android app tip, thank you!

I have been messing A LOT with 32GB SD cards, formatting, writing, re-writing, trying piece by piece instead of full folders, ... After lots of messing, I tried an off brand but 16GB card. This worked straight away with no issues! Really weird, but 16GB was, in my case, the solution i guess. Thanks for all the tips, will continue trying 32GB cards once my daughter gets bored of her current collection.

closed 2023-02-04 by PieterJanSterk