tonuino/TonUINO-TNG #71: compile error

brucetmiller, 2023-03-23:
i am getting the following after trying to compile i am using the wireing as on the website ( classic ) i have changed nothing else

any help would be appretiated

thank you

In file included from C:\Users\bruce\OneDrive\Documents\Arduino\TonUINO-TNG\src\chip_card.hpp:8:0,
                 from C:\Users\bruce\OneDrive\Documents\Arduino\TonUINO-TNG\src\settings.hpp:8,
                 from C:\Users\bruce\OneDrive\Documents\Arduino\TonUINO-TNG\src\tonuino.hpp:4,
                 from C:\Users\bruce\OneDrive\Documents\Arduino\TonUINO-TNG\TonUINO-TNG.ino:1:
C:\Users\bruce\OneDrive\Documents\Arduino\TonUINO-TNG\src\constants.hpp:188:2: error: #endif without #if
 #endif /* SRC_CONSTANTS_HPP_ */
  ^~~~~
In file included from C:\Users\bruce\OneDrive\Documents\Arduino\TonUINO-TNG\src\commands.hpp:7:0,
                 from C:\Users\bruce\OneDrive\Documents\Arduino\TonUINO-TNG\src\tonuino.hpp:5,
                 from C:\Users\bruce\OneDrive\Documents\Arduino\TonUINO-TNG\TonUINO-TNG.ino:1:
C:\Users\bruce\OneDrive\Documents\Arduino\TonUINO-TNG\src\constants.hpp:188:2: error: #endif without #if
 #endif /* SRC_CONSTANTS_HPP_ */
  ^~~~~
In file included from C:\Users\bruce\OneDrive\Documents\Arduino\TonUINO-TNG\src\buttons.hpp:8:0,
                 from C:\Users\bruce\OneDrive\Documents\Arduino\TonUINO-TNG\src\tonuino.hpp:6,
                 from C:\Users\bruce\OneDrive\Documents\Arduino\TonUINO-TNG\TonUINO-TNG.ino:1:
C:\Users\bruce\OneDrive\Documents\Arduino\TonUINO-TNG\src\constants.hpp:188:2: error: #endif without #if
 #endif /* SRC_CONSTANTS_HPP_ */
  ^~~~~
In file included from C:\Users\bruce\OneDrive\Documents\Arduino\TonUINO-TNG\src\serial_input.hpp:7:0,
                 from C:\Users\bruce\OneDrive\Documents\Arduino\TonUINO-TNG\src\tonuino.hpp:7,
                 from C:\Users\bruce\OneDrive\Documents\Arduino\TonUINO-TNG\TonUINO-TNG.ino:1:
C:\Users\bruce\OneDrive\Documents\Arduino\TonUINO-TNG\src\constants.hpp:188:2: error: #endif without #if
 #endif /* SRC_CONSTANTS_HPP_ */
  ^~~~~
In file included from C:\Users\bruce\OneDrive\Documents\Arduino\TonUINO-TNG\src\mp3.hpp:7:0,
                 from C:\Users\bruce\OneDrive\Documents\Arduino\TonUINO-TNG\src\tonuino.hpp:8,
                 from C:\Users\bruce\OneDrive\Documents\Arduino\TonUINO-TNG\TonUINO-TNG.ino:1:
C:\Users\bruce\OneDrive\Documents\Arduino\TonUINO-TNG\src\constants.hpp:188:2: error: #endif without #if
 #endif /* SRC_CONSTANTS_HPP_ */
  ^~~~~
In file included from C:\Users\bruce\OneDrive\Documents\Arduino\TonUINO-TNG\TonUINO-TNG.ino:7:0:
C:\Users\bruce\OneDrive\Documents\Arduino\TonUINO-TNG\src\constants.hpp:188:2: error: #endif without #if
 #endif /* SRC_CONSTANTS_HPP_ */
  ^~~~~

exit status 1

Compilation error: #endif without #if

boerge1, 2023-03-24:
What System you are using, Arduino IDE or platfomio?
There is something wrong with your constants.hpp.
Maybe you did an error while commenting out the line for the classic variant?

boerge1, 2023-03-24:
Please read the Readme. You have to copy the file platform.local.txt into your compiler hw directory.

brucetmiller, 2023-03-24:
that did it, thank you for your help i do not understand german, hopefully it will be easier from here.

closed 2023-03-26 by boerge1