tonuino/TonUINO-TNG #72: Support Rotary Encoder KY-040 for setting the volume

boerge1, 2023-03-30:


Tobi0701, 2023-04-02:
Hallo @boerge1 , 
ich habe deinen [Build ](https://github.com/tonuino/TonUINO-TNG/tree/issue_072 ) ausprobiert, aber leider klappt es nicht so ganz. Der Drehgeber reagiert hauptsächlich nur auf das CLK Signal, wobei der Log so ausschaut als würde er schon das DT Signal erkennen, es aber immer wieder übersteuern. Angeschlossen ist CLK an V+ und DT an V-, GND am GND des V-
Hab ich hier einen Anschluss- oder Denkfehler?

Drehung in Uhrzeigersinn
> 13:42:42.573 -> enter Play
13:42:42.573 -> play 35-118
13:42:45.187 -> Command: 9
13:42:45.187 -> Volume: 25
13:42:45.472 -> Command: 9
13:42:45.472 -> Volume: 25
13:42:45.710 -> Command: 9
13:42:45.710 -> Volume: 25
13:42:46.186 -> Command: 9
13:42:46.186 -> Volume: 25
13:42:46.471 -> Command: 9
13:42:46.471 -> Volume: 25
13:42:46.661 -> Command: 9
13:42:46.661 -> Volume: 25
13:42:46.709 -> Command: 10
13:42:46.709 -> Volume: 24
13:42:46.948 -> Command: 9
13:42:46.948 -> Volume: 25
13:42:46.996 -> Command: 10
13:42:47.044 -> Volume: 24
13:42:47.378 -> Command: 9
13:42:47.378 -> Volume: 25
13:42:47.426 -> Command: 10
13:42:47.474 -> Volume: 24
13:42:47.761 -> Command: 9
13:42:47.761 -> Volume: 25
13:42:47.809 -> Command: 10
13:42:47.857 -> Volume: 24
13:42:48.047 -> Command: 9
13:42:48.047 -> Volume: 25
13:42:48.142 -> Command: 10
13:42:48.142 -> Volume: 24
13:42:48.476 -> Command: 9
13:42:48.476 -> Volume: 25
13:42:48.524 -> Command: 10
13:42:48.571 -> Volume: 24
13:42:49.044 -> Command: 9
13:42:49.044 -> Volume: 25
13:42:49.948 -> Command: 10

Drehung gegen Uhrzeigersinn
> 13:44:10.007 -> enter Play
13:44:10.007 -> play 35-118
13:44:13.047 -> Command: 9
13:44:13.047 -> Volume: 22
13:44:13.381 -> Command: 9
13:44:13.381 -> Volume: 23
13:44:13.714 -> Command: 10
13:44:13.714 -> Volume: 22
13:44:13.762 -> Command: 9
13:44:13.809 -> Volume: 23
13:44:14.233 -> Command: 10
13:44:14.233 -> Volume: 22
13:44:14.279 -> Command: 9
13:44:14.324 -> Volume: 23
13:44:14.609 -> Command: 10
13:44:14.609 -> Volume: 22
13:44:14.657 -> Command: 9
13:44:14.703 -> Volume: 23
13:44:15.085 -> Command: 10
13:44:15.085 -> Volume: 22
13:44:15.133 -> Command: 9
13:44:15.180 -> Volume: 23
13:44:15.419 -> Command: 10
13:44:15.419 -> Volume: 22
13:44:15.467 -> Command: 9
13:44:15.514 -> Volume: 23
13:44:15.751 -> Command: 10
13:44:15.751 -> Volume: 22
13:44:15.799 -> Command: 9
13:44:15.846 -> Volume: 23
13:44:16.132 -> Command: 9
13:44:16.132 -> Volume: 24
13:44:16.513 -> Command: 9
13:44:16.513 -> Volume: 25
13:44:17.135 -> Command: 9
13:44:17.135 -> Volume: 25
13:44:17.278 -> Command: 10
13:44:17.278 -> Volume: 24
13:44:17.326 -> Command: 9
13:44:17.374 -> Volume: 25
13:44:17.518 -> Command: 9
13:44:17.518 -> Volume: 25
13:44:17.708 -> Command: 10
13:44:17.708 -> Volume: 24
13:44:17.756 -> Command: 9
13:44:17.850 -> Volume: 25
13:44:18.040 -> Command: 10
13:44:18.040 -> Volume: 24
13:44:18.087 -> Command: 9
13:44:18.135 -> Volume: 25
13:44:18.418 -> Command: 9
13:44:18.418 -> Volume: 25
13:44:19.133 -> Command: 9
13:44:19.133 -> Volume: 25
13:44:19.227 -> Command: 9
13:44:19.227 -> Volume: 25
13:44:19.370 -> Command: 9
13:44:19.370 -> Volume: 25
13:44:19.653 -> Command: 10
13:44:19.653 -> Volume: 24
13:44:19.701 -> Command: 9
13:44:19.797 -> Volume: 25
13:44:20.035 -> Command: 9
13:44:20.035 -> Volume: 25
13:44:20.464 -> Command: 9
13:44:20.464 -> Volume: 25
13:44:20.846 -> Command: 9
13:44:20.846 -> Volume: 25
13:44:21.227 -> Command: 10
13:44:21.227 -> Volume: 24
13:44:21.274 -> Command: 9
13:44:21.322 -> Volume: 25
13:44:22.706 -> Command: 10
13:44:22.706 -> Volume: 24
13:44:22.754 -> Command: 9
13:44:22.802 -> Volume: 25
13:44:27.035 -> Command: 7
13:44:27.035 -> enter Pause

boerge1, 2023-04-02:
Oh, das war auch noch nicht fertig. Ich hatte in der Bastelkiste noch einen KY040 gefunden, aber der ist wohl anscheinend kaputt. Da kommt bei CLK und DT immer das selbe an.
Du musst CLK an D5 und DT an D6 anschließen. Zusätzlich noch GND an GND. V+ muß nicht angeschlossen werden. Was passiert denn, wenn du den Regler ganz langsam drehst?
Es kann nämlich sein, dass der 50 ms Cycle zu langsam ist. Dann muss ich doch Interrupts nehmen.

Tobi0701, 2023-04-02:
Sorry, da war ich dann wohl zu voreilig mit testen :-) Ich muss gestehen ich suche gerade händeringend auf der AiO Plus Platine nach dem D5, D6 is ja auf das Buttonboard gezogen unter WS(D6), aber wo ist D5 hin?

boerge1, 2023-04-02:
Du kannst auch beliebig andere Inputs nehmen. Du musst dann nur die Datei constants.cpp editieren. Ich hatte es mit der Classic getestet.
Ich warte auf meinen neuen Regler, dann kann ich weitermachen.

boerge1, 2023-04-04:
@Tobi0701 Ich habe das jetzt für die AiOplus mit Interrupts implementiert. 
CLK --> PF2
DT --> PF3
GND --> GND
V+ kann, muss aber nicht angeschlossen werden
Du kannst aber auch beliebig andere freie GPIO nehmen, musst dann nur die Datei constants.hpp editieren. 
Kannst du das mal ausprobieren?

Tobi0701, 2023-04-04:
Leider genau das gleiche Bild, in beide Richtungen nur Command: 9

Tobi0701, 2023-04-04:
nein ohne V+ nur noch GND

Tobi0701, 2023-04-04:
> Hmm, bei mir geht auch PF2 und Pf3.
> 
> > gegen UZS läuft es fast sauber, im UZS "stotterts" etwas, auch wenn ich langsamer drehe
> > Hast du V+ angeschlossen, dann lasse das mal weg.

... klassischer Flüchtigkeitsfehler, man muss schon PE und PF unterscheiden können...

du hast recht, es funktioniert auch an PF2 und PF3

boerge1, 2023-04-04:
>nein ohne V+ nur noch GND

Dann versuche mal mit V+, ob sich da was ändert. 
Bei mir war es eigentlich recht stabil. 
Ist aber auch noch nicht ganz fertig. Ich muss bei den Interrupts noch was optimieren.

boerge1, 2023-04-04:
Created pull request #77

boerge1, 2023-04-06:
@Tobi0701 ich habe jetzt die finale Version hochgeladen. Kannst du das nochmal testen. 
Du musst jetzt dabei die Datei constants.hpp editieren, das Feature soll ja nur für diejenigen sein, die es auch haben wollen. Auch die Pins musst du editieren, Standard sind jetzt 

PE1 | 31 / A9 | GPIO | Rotary Encoder Buchse CLK
PE2 | 32 / A10 | GPIO | Rotary Encoder Buchse DT

wie in [www.leiterkartenpiraten.de/tonuino-all-in-one-plus-quick-start-guide/](https://www.leiterkartenpiraten.de/tonuino-all-in-one-plus-quick-start-guide/) beschrieben.

Tobi0701, 2023-04-06:
Hallo @boerge1 , ich teste es heute Abend. Was mir aber direkt auffällt ist das PE1 und PW2 zwar prinzipiell für den Drehgeber vorbereitet sind, aber man dafür einen 5 poligen Stecker brauch, da sind die PINS PF2 und PF3 die bessere "Standard" Wahl mMn.

boerge1, 2023-04-07:
Merged pull request #77

closed 2023-04-07 by boerge1