tonuino/TonUINO-TNG #73: Support LED or NeoPixel Ring

boerge1, 2023-03-30:
Dem Wunsch schließe ich mich an. Am besten und flexibelsten wäre es über adressierbare LEDs, also Neopixel, WS2812b und Konsoren.
Aktuell gibt es hauptsächlich den Affenbox-Fork mit direkter Versorgung der LED über WS mit Pulsieren bei Pause durch PWM oder den Always-Quene-Fork mit Pixelring (aber mir persönlich zu wilder Animation).

Cool wäre ein Zwischending, also z.B. eine entspanntere Ring-Animation oder eine einzelne LED (also Ring/Stripe mit n=1 LEDs), die unterschiedliche Zustände anzeigen kann.
Dafür könnte man zum Beispiel eine solche Zuordnung treffen:

    Tonuino startet und ist noch nicht bereit, eine Karte aufzulegen: rotes Pulsieren/Drehen bzw. bei Einzel-LED wechelnder Übergang rot/weiß
    Tonuino ist hochgefahren und bereit für eine Karte: grünes Pulsieren bzw. wechelnder Übergang grün/weiß
    Pause: blaues Pulsieren bzw. wechelnder Übergang blau/weiß
    Tonuino spielt ab: bunter Farbwechsel; insbesondere bei Ring oder Stripe nicht zu wild, eher ganzen Ring in einer Farbe und beginnend von LED1 dann neue Farbe bis alle in der neuen Farbe sind und dann weiter mit der nächsten Farbe; beim Stripe evtl eine Farbe hin, nächste Farbe zurück
    nur bei Ring oder Stripe: Lautstärke wird im Verhältnis (VolMax - VolMin / Anzahl LEDs) angezeigt, ggf. in rot wenn leiser gestellt wird und in grün wenn lauter gestellt wird
    Defines für Typ (Ring oder Stripe), Anzahl LEDs, Offset für gedreht eingebaute Ringe, ggf. Animationsgeschwindigkeit

TheMBeat, 2023-06-05:
Das wäre ein cooles Feature.
Gibt es schon ein Plan wann es umgesetzt wird?

boerge1, 2023-06-08:
Ich werde das als nächtes angehen. Auf jeden Fall wird zuerst der Ring implementiert, da ich dazu auch die HW habe. 
Natürlich wir das nur was für die AiOplus, bei den anderen Varianten ist der Speicher schon ausgereizt.

TheMBeat, 2023-06-08:
Cool, dass es als nächstes ansteht, dann muss ich mich nicht reinwuchsen. Aber schlecht das es nur für den AiO+ implementiert wird. 
Ich habe leider nur den AiO, kann nicht etwas anderes rausgenommen werden? Z.B das Logging um Platz zu schaffen?

TheMBeat, 2023-06-09:
Hey @boerge1, wenn ich den Branch 73 verwende und SerialInput auskommentier, passt die Firmware auf den AiO (99,7% benutzt).
Die IDLE Animation ist bei mir sehr wild, da ich nur 12 LED habe, ein pulsieren ist wahrscheinlich angenehmer.
Allerdings reagiert der RFID Reader nicht, selber Aufbau mit dem Always-Cue-Fork funktioniert, da werden Karten erkannt. Hast du eine Idee warum?

Ich habe den SD-Karten Inhalt aus dem Branch genommen, bekomme aber einen Fehler mit der Datei 131, obwohl sie da ist.

boerge1, 2023-06-10:
Ja, die Animationen sind natürlich noch nicht fertig. Es ging erstmal darum, daß überhaupt zu integrieren.
Wenn du sonstige Probleme hast, dann versuche doch erstmal den Main Branch von der 3.1.
Wenn es da auch nicht geht, dann schreibe das doch bitte hier https://discourse.voss.earth/t/tester-gesucht-tonuino-3-1. Und ein Log wäre auf jeden Fall hilfreich.

TheMBeat, 2023-06-10:
Hey @boerge1, danke für deine Antwort. 

Ich werde mal den Main-Branch testen und bei Probleme dan im Forum schreiben inkl. LOG.

Das es nicht die entgültigen Animationen sind haben ich mir schon. Gedacht, wollte trotzdem ein Feedback geben.

boerge1, 2023-06-14:
@TheMBeat siehe [hier](https://discourse.voss.earth/t/neopixel-ring-fuer-die-version-3-1/13072)
Ich habe die Animationen verbessert.

TheMBeat, 2023-06-14:
Ich schaue mal rein,

Mit dem Main-Branch hatte ich Kant kurz Erfolg mit Dem RFID-Leser.
Dann ging es nicht mehr. 

Als wenn er nicht richtig initialisiert wird.

boerge1, 2023-08-02:
Created pull request #98

boerge1, 2023-09-03:
pull request #98 merged

closed 2023-09-03 by boerge1