tonuino/TonUINO-TNG #76: <<Phopp>> sound at poweroff

grandmastermarco, 2023-04-03:
I have a loud sound when powering off.
Is there a possibility to adjust timing of powering off the AMP similar to this guide?

[https://discourse.voss.earth/t/aio-knacken-des-lautsprechers-beim-ein-und-ausschalten/10305/2]

thanks in advance :)

boerge1, 2023-04-04:
@grandmastermarco I added a delay, but this is only for AiO or AiOplus
Could you please check the branch issue_076?

grandmastermarco, 2023-04-04:
hi! thanks for the quick help.
i have the AiO (non-plus version), therefore delay could help.

when trying out firmware 3.1.3 i used the online firmware updater without a problem.
now to try branch issue_076, i get an error when compiling in AruinoIDE.
Any chance you can offer troubleshooting support? seems like a problem with DF_Player_Mini.

I appreciate the help!

```
`Arduino: 1.8.19 (Windows 10), Board: "LGT8F328, Default (64), External (assumes 32MHz crystal), 16 MHz, 328P-LQFP32 (e.g. MiniEVB nano-style or WAVGAT)"

In file included from C:\Users\marco\Tonuino\Firmwares\TonUINO 3.1.3\TonUINO-TNG\src\mp3.hpp:5:0,

                 from C:\Users\marco\Tonuino\Firmwares\TonUINO 3.1.3\TonUINO-TNG\src\tonuino.hpp:9,

                 from C:\Users\marco\Tonuino\Firmwares\TonUINO 3.1.3\TonUINO-TNG\TonUINO-TNG.ino:1:

C:\Users\marco\Documents\Arduino\libraries\DFPlayer_Mini_Mp3_by_Makuna\src/DFMiniMp3.h: In instantiation of 'uint16_t DFMiniMp3<T_SERIAL_METHOD, T_NOTIFICATION_METHOD>::listenForReply(uint8_t) [with T_SERIAL_METHOD = SoftwareSerial; T_NOTIFICATION_METHOD = Mp3Notify; uint16_t = unsigned int; uint8_t = unsigned char]':

C:\Users\marco\Documents\Arduino\libraries\DFPlayer_Mini_Mp3_by_Makuna\src/DFMiniMp3.h:105:27:   required from 'void DFMiniMp3<T_SERIAL_METHOD, T_NOTIFICATION_METHOD>::loop() [with T_SERIAL_METHOD = SoftwareSerial; T_NOTIFICATION_METHOD = Mp3Notify]'

C:\Users\marco\Documents\Arduino\libraries\DFPlayer_Mini_Mp3_by_Makuna\src/DFMiniMp3.h:429:17:   required from 'void DFMiniMp3<T_SERIAL_METHOD, T_NOTIFICATION_METHOD>::sendPacket(uint8_t, uint16_t, uint16_t) [with T_SERIAL_METHOD = SoftwareSerial; T_NOTIFICATION_METHOD = Mp3Notify; uint8_t = unsigned char; uint16_t = unsigned int]'

C:\Users\marco\Documents\Arduino\libraries\DFPlayer_Mini_Mp3_by_Makuna\src/DFMiniMp3.h:282:19:   required from 'void DFMiniMp3<T_SERIAL_METHOD, T_NOTIFICATION_METHOD>::start() [with T_SERIAL_METHOD = SoftwareSerial; T_NOTIFICATION_METHOD = Mp3Notify]'

C:\Users\marco\Tonuino\Firmwares\TonUINO 3.1.3\TonUINO-TNG\src\mp3.hpp:182:41:   required from here

C:\Users\marco\Documents\Arduino\libraries\DFPlayer_Mini_Mp3_by_Makuna\src/DFMiniMp3.h:511:62: error: no matching function for call to 'Mp3Notify::OnPlayFinished(DfMp3_PlaySources, uint16_t&)'

                         T_NOTIFICATION_METHOD::OnPlayFinished(DfMp3_PlaySources_Usb, replyArg);

                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

In file included from C:\Users\marco\Tonuino\Firmwares\TonUINO 3.1.3\TonUINO-TNG\src\tonuino.hpp:9:0,

                 from C:\Users\marco\Tonuino\Firmwares\TonUINO 3.1.3\TonUINO-TNG\TonUINO-TNG.ino:1:

C:\Users\marco\Tonuino\Firmwares\TonUINO 3.1.3\TonUINO-TNG\src\mp3.hpp:140:15: note: candidate: static void Mp3Notify::OnPlayFinished(DfMp3&, DfMp3_PlaySources, uint16_t)

   static void OnPlayFinished      (DfMp3&, DfMp3_PlaySources source, uint16_t track);

               ^~~~~~~~~~~~~~

C:\Users\marco\Tonuino\Firmwares\TonUINO 3.1.3\TonUINO-TNG\src\mp3.hpp:140:15: note:   candidate expects 3 arguments, 2 provided

In file included from C:\Users\marco\Tonuino\Firmwares\TonUINO 3.1.3\TonUINO-TNG\src\mp3.hpp:5:0,

                 from C:\Users\marco\Tonuino\Firmwares\TonUINO 3.1.3\TonUINO-TNG\src\tonuino.hpp:9,

                 from C:\Users\marco\Tonuino\Firmwares\TonUINO 3.1.3\TonUINO-TNG\TonUINO-TNG.ino:1:

C:\Users\marco\Documents\Arduino\libraries\DFPlayer_Mini_Mp3_by_Makuna\src/DFMiniMp3.h:515:62: error: no matching function for call to 'Mp3Notify::OnPlayFinished(DfMp3_PlaySources, uint16_t&)'

                         T_NOTIFICATION_METHOD::OnPlayFinished(DfMp3_PlaySources_Sd, replyArg);

                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

In file included from C:\Users\marco\Tonuino\Firmwares\TonUINO 3.1.3\TonUINO-TNG\src\tonuino.hpp:9:0,

                 from C:\Users\marco\Tonuino\Firmwares\TonUINO 3.1.3\TonUINO-TNG\TonUINO-TNG.ino:1:

C:\Users\marco\Tonuino\Firmwares\TonUINO 3.1.3\TonUINO-TNG\src\mp3.hpp:140:15: note: candidate: static void Mp3Notify::OnPlayFinished(DfMp3&, DfMp3_PlaySources, uint16_t)

   static void OnPlayFinished      (DfMp3&, DfMp3_PlaySources source, uint16_t track);

               ^~~~~~~~~~~~~~

C:\Users\marco\Tonuino\Firmwares\TonUINO 3.1.3\TonUINO-TNG\src\mp3.hpp:140:15: note:   candidate expects 3 arguments, 2 provided

[... 128 lines cut ...]
                     T_NOTIFICATION_METHOD::OnError(replyArg);

                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~

In file included from C:\Users\marco\Tonuino\Firmwares\TonUINO 3.1.3\TonUINO-TNG\src\tonuino.hpp:9:0,

                 from C:\Users\marco\Tonuino\Firmwares\TonUINO 3.1.3\TonUINO-TNG\TonUINO-TNG.ino:1:

C:\Users\marco\Tonuino\Firmwares\TonUINO 3.1.3\TonUINO-TNG\src\mp3.hpp:139:15: note: candidate: static void Mp3Notify::OnError(DfMp3&, uint16_t)

   static void OnError             (DfMp3&, uint16_t errorCode);

               ^~~~~~~

C:\Users\marco\Tonuino\Firmwares\TonUINO 3.1.3\TonUINO-TNG\src\mp3.hpp:139:15: note:   candidate expects 2 arguments, 1 provided

exit status 1

Fehler beim Kompilieren für das Board LGT8F328.

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.
`
```

grandmastermarco, 2023-04-04:
branch issue_076 fixed the very loud phopp sound at poweroff. thank you very much!

closed 2023-04-04 by grandmastermarco