tonuino/TonUINO-TNG #97: DfPlayer Error: 1

nanu-c, 2023-07-20:
Hey, 

is this an hardware error? I have the TonUINO All-In-One board.
Regarding the df player library `DfPlayer Error: 1` means busy, maybe the sd card isn't recognized?

Here is the complete log:

```
 _____         _____ _____ _____ _____
|_   _|___ ___|  |  |     |   | |     |
  | | | . |   |  |  |-   -| | | |  |  |
  |_| |___|_|_|_____|_____|_|___|_____|
TonUINO Version 3.1 - refactored by Boerge1

created by Thorsten Voß and licensed under GNU/GPL.
Information and contribution at https://tonuino.de.

V3.1.3 16.07.23

Ver:2
Vol:25 5 15
EQ:1
LOC:0
ST:0
IB:1
AL:0
AP:1111
PCR:0
Volume: 15
DfPlayer Error: 1
enter Idle
Command: 6
enqueue mp3 262 0
play mp3 262
DfPlayer Error: 1
DfPlayer Error: 1
missing OnPlayFinished
Card Inserted
CardData:  31 73 3b 74 02 01 05 00 00
enter StartPlay
enqueue mp3 262 0
play mp3 262
DfPlayer Error: 1
Card Removed
missing OnPlayFinished
DfPlayer Error: 1
0 tr in folder 1
enter Play
DfPlayer Error: 1
enter Idle
```

boerge1, 2023-07-22:
What version of DFMiniMp3 library you are using?

nanu-c, 2023-07-22:
i am using the version 1.1.0
[image]

boerge1, 2023-07-22:
Is It a very new AIO hat never run before?
I would try the following
- remove the SD Card and start. If the same error comes, the it should bei an HW error and you should send an Email to support@leiterkartenpiraten.de
- rewrite the SD Card (does it have the right directory structure and filenames?)
- try another SD card

boerge1, 2023-07-29:
@nanu-c  Are There any news?

boerge1, 2023-08-10:
Seams, it is not an software error.

closed 2023-08-10 by boerge1