tonuino/TonUINO-TNG #99: Fix Doku in Readme.md - DFPlayer Mini Mp3 v1.1.0 not workin

tschambe, 2023-08-22:
Hi everyone,

first of all thanks for your awesome work.
The version of the library "DFPlayer Mini Mp3 by Makuna" mentioned in the README.md (with 1.1.0) does not work with the current firmware.
I spent quite some time today with debugging this but a switch to the most recent version 1.2.1 fixed all my problems.
Maybe it is worth checking if this problem also occurs in your setup and then modify the Readme accordingly.
Thank again and all the best,
Thomas

boerge1, 2023-08-22:
What df player you are using? Please have a look at the chip.
We tried version 1.2.0 and there were problems. I will test now the version 1.2.1

boerge1, 2023-08-23:
Tested version 1.2.1 with my HW. 

Classic: runs without error, but also version 1.1.0 was running without any error
AiO plus: has problems playing tracks in the mp3 folder, with version 1.1.0 all is running without any error

Will ask other in the forum [TonUINO](https://discourse.voss.earth/)
Until clarification remain on version 1.1.0

boerge1, 2023-08-25:
@tschambe Could you please test the branch issue_099 with your DFPlayer? (Of course with the library version 1.2.1)
I did a change that this new version is also working with the other DFPlayer that are used for TonUINO.

tschambe, 2023-08-25:
Yes I will do this on Monday (I don't have access to the hardware right now...). Thanks for the fix!

tschambe, 2023-08-29:
So I finally checked it. Your fix works great with my hardware. Library Version 1.2.1

closed 2023-09-04 by boerge1