tonuino/TonUINO-TNG #110: Occasional ''Auth failed: 1

krylovsk, 2023-09-18:
Card Inserted
Auth  failed: 1
Card Removed

boerge1, 2023-09-19:
Normally this error (if not permanent) occurs when the card is held too close or too far away from the reader. Or if the voltage on the reader is to low. Do you use a NANO or AiO the reader is connected to? 
It is better to discuss this [here](https://discourse.voss.earth/) as it seems to be an HW issue. Although this is a German forum, you can also ask in English.

krylovsk, 2023-09-20:
I'm using NANO and looks like it got better once I fixed the reader pcb in the case. 
I just got confused by the error, but looking in the code now I understand it's the first operation that actually attempts to read data from the card. Will close the issue and raise on the forum if I'll keep running into it.

closed 2023-09-20 by krylovsk