tonuino/TonUINO-TNG #114: Installation - instructions missing

Eheran1, 2023-10-02:
> [Installation](https://github.com/tonuino/TonUINO-TNG/tree/main#installation)
> Die SD Karte (Ordner mp3 und advert) hat sich gegenüber der Version 3.1.2 geändert. Hier kann man die Dateien downloaden: [tonuino.github.io/TonUINO-TNG/sd-card.zip](https://tonuino.github.io/TonUINO-TNG/sd-card.zip)

This is all of the information provided on how to install this library here on github. It can not be found in the Arduino Library manager (Arduino IDE 2.2.1). It can also not be installed via downloading .zip -> add .zip library -> 
> Error: 13 INTERNAL: Library install failed: moving extracted archive to destination dir: library not valid
I assume this is because of the folder structure not matching what Arduino expects.

> Allgemeine Anleitungen zum Einrichten der IDE findet man hier [www.tonuino.de/TNG](https://www.tonuino.de/TNG) und hier [www.leiterkartenpiraten.de](https://www.leiterkartenpiraten.de/)

The first link is outdated and links to the old TonUINO. The second link includes information how to add the board in the board manager but not about the library - or I was unable to locate that information.

So the question is: How to install and/or use this library?

GuteLaune, 2023-10-02:
> The first link is outdated and links to the old TonUINO.

Why do you think so? The site looks similar, but the version you get there is this one.

Eheran1, 2023-10-02:
Go to www.tonuino.de/TNG -> "Anleitung" -> 
>Firmware aufspielen
>Nun müsst ihr den Sketch TonUINO.ino aus meinem GitHub Repository mit der Arduino IDE öffnen. [= old repository with one massive .ino file, 1800 lines]

>Quellcode
>Der Sourcecode steht auf meinem [GitHub Repository](https://github.com/xfjx/TonUINO) bereit [= old repository that then links to this one]

Alternatively, go to https://www.tonuino.de/TNG/ -> "Source Code" -> directly links to this respository

In any case, the actual information how to use this repository is nowhere to be found. The installation section is only about the sound-effects(?) for the SD-card.

I assume I can copy all the .ino etc. files into a sketch folder and use it that way, but I doubt that is the way it is supposed to be given the high quality of the whole project. Hence I want to give feedback as to what I found to be missing.

boerge1, 2023-10-03:
This is not a library but the Software for the so called TonUINO project.
You can compile and upload this project with Platformio or Arduino IDE. For this, of course, you have to clone or download this repository and to use it as project directory.
And as main requirement you have to have the hardware for this project. Do you have it?

Eheran1, 2023-10-03:
I have the parts sold in the "AZDelivery TonUINO Set", except that I bought them individually and will use a ESP32 for BT functionalities I might want to implement at some point.

boerge1, 2023-10-03:
This software doesn't work with esp32.

GuteLaune, 2023-10-03:
> will use a ESP32 for BT functionalities I might want to implement at some point.

Maybe you want to use an esp32 based project like https://github.com/biologist79/ESPuino then.

boerge1, 2023-10-04:
> Let us assume I am a beginner with Arduino. I install everything, follow the links here and now how do I actually use this repository?

Simply compile and upload this to the appropriate hardware and enjoy.
All other is described in the links.

closed 2023-10-06 by Eheran1