tonuino/TonUINO-TNG #157: Error when compiling with arduino-cli for AiO+

dosas, 2024-01-03:
```
arduino-cli compile --fqbn arduino:avr:nano --build-property "compiler.cpp.extra_flags=-std=gnu++17"
```

yields

```
In file included from /home/ich/TonUINO-TNG/src/chip_card.hpp:8:0,
                 from /home/ich/TonUINO-TNG/src/settings.hpp:8,
                 from /home/ich/TonUINO-TNG/src/tonuino.hpp:4,
                 from /home/ich/TonUINO-TNG/TonUINO-TNG.ino:1:
/home/ich/TonUINO-TNG/src/constants.hpp:65:41: error: 'A14' was not declared in this scope
 inline constexpr uint8_t   potiPin    = A14; // AiO+ PF4
                                         ^~~
/home/ich/TonUINO-TNG/src/constants.hpp:65:41: note: suggested alternative: 'A4'
 inline constexpr uint8_t   potiPin    = A14; // AiO+ PF4
                                         ^~~
                                         A4
/home/ich/TonUINO-TNG/src/constants.hpp:248:60: error: 'Serial3' was not declared in this scope
 inline constexpr HardwareSerial &dfPlayer_serial         = Serial3;
                                                            ^~~~~~~
/home/ich/TonUINO-TNG/src/constants.hpp:248:60: note: suggested alternative: 'Serial'
 inline constexpr HardwareSerial &dfPlayer_serial         = Serial3;
                                                            ^~~~~~~
                                                            Serial

Used library                Version Path
EEPROM                      2.0     /home/ich/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/EEPROM
MFRC522                     1.4.10  /home/ich/Arduino/libraries/MFRC522
SPI                         1.0     /home/ich/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI
JC_Button                   2.1.2   /home/ich/Arduino/libraries/JC_Button
DFPlayer Mini Mp3 by Makuna 1.2.3   /home/ich/Arduino/libraries/DFPlayer_Mini_Mp3_by_Makuna

Used platform Version Path
arduino:avr   1.8.6   /home/ich/.arduino15/packages/arduino/hardware/avr/1.8.6
Error during build: exit status 1

```

It works fine when compiling for classic (uncomment define directive)

* Are the suggestions made by the compiler correct? (If I apply them it compiles)
* As far as I know the AiO+ also uses the arduino nano?

boerge1, 2024-01-03:
The AiO+ doesn't have a nano as processor. Please refer the Links in the Readme for instructions how to configure the Arduino IDE for the AiO+.
Do you really own an AiO+ from [here](https://www.leiterkartenpiraten.de/)?

dosas, 2024-01-05:
For future reference:

## AiO
https://www.leiterkartenpiraten.de/tonuino-all-in-one-die-arduino-ide-einrichten/
Arduino Uno (`--fqbn arduino:avr:uno`)

## AiO+
https://www.leiterkartenpiraten.de/tonuino-all-in-one-plus-die-arduino-ide-einrichten/
Arduino Yun (`--fqbn arduino:avr:yun`)

Still getting the same errors for the AiO+ when using 'yun' but since it is discontinued it does not make sense to follow up

## Every
`--fqbn arduino:megaavr:nona4809`

dosas, 2024-01-05:
Thanks. I tried that too after finding it in the .github actions configuration files.

It should just work by using [`pio run`](https://github.com/tonuino/TonUINO-TNG/blob/main/.github/workflows/build_and_run_tests.yml#L47)?

```
Environment          Status    Duration
-------------------  --------  ------------
TonUINO_Classic_3    SUCCESS   00:00:56.063
TonUINO_Classic_5    SUCCESS   00:00:05.401
TonUINO_Classic_3x3  SUCCESS   00:00:05.341
TonUINO_Every_3      FAILED    00:00:07.466
TonUINO_Every_5      FAILED    00:00:02.513
TonUINO_Every_3x3    FAILED    00:00:02.674
ALLinONE             FAILED    00:00:07.993
ALLinONE_3x3         FAILED    00:00:02.550
ALLinONE_Plus        FAILED    00:00:05.265
ALLinONE_Plus_3x3    FAILED    00:00:02.570
```

closed 2024-01-05 by dosas