tonuino/TonUINO-TNG #159: 3 Achs Gyroskop, GY-521

Max23888, 2024-01-18:
Hello, is it possible to implement a 3 axis Gyroskop for next track and previous track? 

Thank you very much
Best Regards 
Markus

boerge1, 2024-01-21:
Maybe it is possible for the AiOplus and for the Nano Every. I'm going to have a look at it.

boerge1, 2024-02-03:
@Max23888 could you please test it with the branch [issue_159](https://github.com/tonuino/TonUINO-TNG/tree/issue_159)? 

You have to connect the SDA and SCL pins with the corresponding pins of the board:
AiOplus: PA2 and PA3
Classic: A4 and A5

Additionally you can connect the INT pin with PB2 (AiOplus) or D8 (Nano Every), otherwise you have to disable the MPU6050_INT define in constants.hpp. But I have better results with the interrupt. 

The detection is made on x-axis and the pins on the GY-521 have to be on the right side. 

I'm not fully satisfied with the detection and will further work on it.

boerge1, 2024-04-10:
pull request #192 created