tonuino/TonUINO-TNG #173: Improve Rorary Encoder implementation (also for UNO and optional for next/previous)

boerge1, 2024-02-18:
* if interrupt pin is not supported (UNO) use a timer ISR (200 Hz) with polling
* optional feature: if one click is done in one second interpret it as next/previous

Thanks @DeKi90 for the idea

boerge1, 2024-02-18:
pull request #174 created

DeKi90, 2024-02-18:
Thanks @boerge1.
I didn't spot the existing timer implementation, that is great.
Can you clarify why you chose to make the volume and long_press timeouts independent?

boerge1, 2024-02-18:
Because they have different meaning. The vol timer is to show, that the last change was a vol change and until it expires, every change should also a vol change (also if pos ==1). So after it expires there will be no remaining pos! =0) and no up/down cmd will be raised afterwards without turning the rot encoder.
With this implementation we have no wait vor the vol cmd, right after two clicks the vol will be changed.

DeKi90, 2024-02-19:
Ah.. that makes sense, yes.
My approach was to handle this the same way as the buttons to avoid small users jamming the volume up all the time. But most likely this input is for more dexterous users in the first place :-D

boerge1, 2024-02-24:
@DeKi90 did you already test my implementation?

boerge1, 2024-03-09:
pull request #174 merged

closed 2024-03-09 by boerge1