tonuino/TonUINO-TNG #209: "Kitamodus" Modifikationskarte nicht kompatibel mit "Stopp wenn Karte weg"

Simson002, 2024-06-15:
Hallo, 
mir ist aufgefallen, dass trotz der Modifikationskarte "Kitamodus" neue Lieder abgespielt werden können wenn man die Funktion "Stopp wenn Karte weg" aktiviert hat. Außerdem werden Lieder auch nicht zu Ende gespielt. 

Vorgehen: 
- Modifikationskarte Kitamodus auflegen
- Karte auflegen 
- Karte wegnehmen (Lied wird unterbrochen trotz "Kitamodus")
- neue Karte auflegen --> neues Lied wird gespielt. 

Die andere Funktion, dass die Vor- und Zurück - Tasten gesperrt sind funktioniert ganz normal; für Alben oder Playlists kann man also erzwingen dass nicht "vorgespult" werden kann. 
Aber die Intention, dass Lieder in jedem Fall zu Ende gespielt werden, das klappt nicht.

boerge1, 2024-06-16:
Ja, der Kita Modus ist im Moment nicht kompatibel mit "Pause wenn Karte weg".
Wie soll sich denn der TonUINO verhalten, wenn die Karte weggenommen wird und eine andere aufgelegt wird?

Simson002, 2024-06-16:
Hmm, für mich am sinnvollsten wäre, wenn der Kita-Modus aktiv ist: 

Wenn man eine Karte startet, wird sie einfach in jedem Fall zu Ende gespielt, also auch, wenn man die Karte entfernt. 
Legt man eine neue Karte auf, während das alte Lied noch spielt, sollte entweder gar nichts passieren, oder die neue Karte sollte in eine Art Queue kommen und spielen, sobald das laufende Lied zu Ende ist. 

ABER: ich finde es auch völlig in Ordnung, wenn die beiden Modi eben nicht kompatibel sind, ich hab überhaupt keine Ahnung, wie schwierig sowas im Code umzusetzen wäre. 

Ich wollte nur darauf hinweisen, da es mir heute aufgefallen ist, und weil ich nicht wusste, ob sowas früher schonmal möglich gemacht wurde. 

Viele Grüße!

PS Ist so ein Hinweis eigentlich hier sinnvoller oder lieber drüben im Forum bei voss.earth?

boerge1, 2024-06-17:
Im Prinzip ist es dann mit dem Kita Modus so, dass "Pause wenn Karte weg" ausgeschaltet ist. 
Ich schau mir das dann mal an, habe aber gerade Urlaub.
Im Forum ist sowas wohl besser aufgehoben.

boerge1, 2024-06-28:
Ich habe mir das jetzt im Code angeschaut. Es ist nicht so einfach möglich, "Pause wenn Karte weg" per Modifier auszuschalten. Dafür fehlt noch der Callback für die Card-Events. Es würde aber auch Flash Programm-Speicher benötigen, wenn ich diese virtuelle Funktion hinzufüge. Das Einzige, was einfach ginge, wäre die Pause generell im Kita-Modus auszuschalten. Aber ist vielleicht auch nicht so gut, wenn die Pause Taste da nicht mehr geht. 
Ich würde es dann lieber so lassen. Ist das für dich ok? Ansonsten würde ich im Forum mal ein Thread aufmachen.

Simson002, 2024-06-28:
Danke fürs drüber schauen!
Nein, ist ja völlig okay wenn zwei, jeweils auf ihre Art tolle Funktionen nicht kompatibel sind.

Ein winziger Vermerk im Readme reicht völlig aus.
Oder, was auch cool wäre, ein Nachsatz im Admin-Menü (beim Erstellen der Modifikationskarte vielleicht à la: "Stopp wenn Karte weg Funktion aktivieren, Hinweis, nicht kompatibel mit Kita-Modus).
Grüße und Danke, das Issue kann geschlossen werden :-)

closed 2024-07-01 by boerge1