tonuino/TonUINO-TNG #244: Compilation error: 'maxTracksInFolder' was not declared in this scope

scrapix, 2024-11-04:
Hallo zusammen, 

leider habe ich nach dem befolgen der Anleitung sowohl auf Mac OSX ARM als auch auf einer alten Windows Maschine den gleichen folgenden Fehler bereits beim Kompilieren in der Arduino IDE. 

Hat jemand eine Idee, wie ich das beheben kann?

Vielen Dank!

``` 
In file included from /Users/i555478/dev/TonUINO-TNG/src/tonuino.hpp:11:0,
                 from /Users/i555478/dev/TonUINO-TNG/TonUINO-TNG.ino:1:
/Users/i555478/dev/TonUINO-TNG/src/mp3.hpp:265:26: error: 'maxTracksInFolder' was not declared in this scope
   typedef queue<uint8_t, maxTracksInFolder> track_queue;
                          ^~~~~~~~~~~~~~~~~
/Users/i555478/dev/TonUINO-TNG/src/mp3.hpp:265:43: error: template argument 2 is invalid
   typedef queue<uint8_t, maxTracksInFolder> track_queue;
                                           ^
/Users/i555478/dev/TonUINO-TNG/src/mp3.hpp: In member function 'uint8_t Mp3::getCurrentTrack()':
/Users/i555478/dev/TonUINO-TNG/src/mp3.hpp:232:50: error: request for member 'get' in '((Mp3*)this)->Mp3::q', which is of non-class type 'Mp3::track_queue {aka int}'
   uint8_t getCurrentTrack() { return playing ? q.get(current_track) : 0; }
                                                  ^~~

exit status 1

Compilation error: 'maxTracksInFolder' was not declared in this scope
```

boerge1, 2024-11-05:
Hast du vielleicht vergessen, die Datei constants.hpp zu editieren und eine Variante zu aktivieren (classic, aio oder aioplus)?

scrapix, 2024-11-06:
Vielen Dank für den Hinweis! Ich habe nun die VSCODE Integration ausprobiert und damit hat es funktioniert. Ich nehme an, dass beim Auswählen der Konfigurationsvariante, die constants.hpp automatisch angepasst wird. Zuvor hatte ich den Schritt tatsächlich übersehen.

closed 2024-11-06 by scrapix