tonuino/TonUINO-TNG #245: Support NFC 215 Tags

scrapix, 2024-11-06:
Hallo zusammen, 

ich habe heute versucht [NFC 215 Tags](https://www.amazon.de/gp/product/B08CRMH9KM/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) zu beschreiben. Leider konnte der RC522 nichts damit anfangen.

Wäre es möglich den Support dieser Tags in das TonUINO Projekt mit aufzunehmen?

Laut folgendem Kommentar kann der RC522 grundsätzlich die NFC215 Tags lesen.
https://forum.arduino.cc/t/reading-writing-ntag215-with-rc522/1111130

Diese Tags scheinen deutlich kompatibler zu sein und z.B. auch über iOS NFC Tools beschrieben werden können.

boerge1, 2024-11-07:
@scrapix: Kannst du mal die Software vom Branch "issue_245" laden und das Log hier posten, wenn du die Karte auflegst?

scrapix, 2024-11-07:
@boerge1 Leider keine Ausgabe im Serial Monitor. Nur die TonUINO Tags funktionieren. Hmm.. Vielleicht findet sich ein anderer Weg die ganzen Tags zu beschreiben 😊 
Vielen Dank für das testen!

boerge1, 2024-11-08:
Nanu, bekommst du gar keine Log Ausgaben oder nur nicht, wenn du den Tag auflegst?

scrapix, 2024-11-08:
@boerge1 ich bekomme keinen Log im Serial Output. Es scheint als würde RC522 nichts erkennen.
Morgen kommen neue [Ntag215](https://www.amazon.de/dp/B088RHZ53P?ref=ppx_yo2ov_dt_b_fed_asin_title). Ich möchte prüfen ob diese erkannt werden.

boerge1, 2024-11-25:
Also ist der RC522 nicht kompatibel mit Ntag215? Ich würde dann das Issue schließen.

scrapix, 2024-11-27:
@boerge1 Ich denke, dass im TONuino Code der Support für NTAG215 implementiert werden müsste, sodass es funktioniert. Leider sind meine (HW) Programmierskills auf niedrigem Niveau und ich kann es nicht selbst implementieren zum testen.
Laut meiner online Recherche müsste es grundsätzlich jedoch funktionieren.

- Funktionsbestätigung: https://regnerischernachmittag.wordpress.com/tag/rc522/
- Implementierungscode: https://forum.arduino.cc/t/reading-writing-ntag215-with-rc522/1111130

boerge1, 2024-11-27:
Beim Erkennen einer Karte unterscheidet sich der Code vom TonUINO aber doch gar nicht vom obigen Beispiel Code. Und es hätte im Log wenigstens eine Nachricht kommen müssen, wenn eine Karte aufgelegt wird. Und wenn diese nicht kommt nützt es auch nichts, wenn ich den Code zum Lesen ändere.
Du kannst mir aber sehr gerne so eine Karte schicken und ich kann das genauer untersuchen. Schicke mir am besten im [Forum](https://discourse.voss.earth/ ) eine PM.

scrapix, 2024-11-27:
@boerge1 Ein Italiener beschreibt in einem [Video](https://www.youtube.com/watch?v=d0BxIG5Kfnc) ebenfalls, dass und wie es geht.
Vielen Dank für deine Unterstützung, selbstverständlich sende ich dir gerne ein paar N215 Tags!

boerge1, 2025-01-13:
pull request #256 merged

closed 2025-01-13 by boerge1