tonuino/TonUINO-TNG #255: Arduino-IDE, V3.2.1_2024.11.23, CPP Type error

ekacoei, 2024-12-24:
I do not know how to compile the current firmware.

I am using tag `V3.2.1_2024.11.23` and `arduino-ide_2.3.4_Linux_64bit.AppImage`, my compiler supports
```
g++ --std=c++17 /dev/null
```
yet while compiling with Arduino IDE I get 
```
~/tonuino/TonUINO-TNG/src/chip_card.cpp:51:66: error: 'printStatusCode' function uses 'auto' type specifier without trailing return type
 auto printStatusCode(MFRC522& mfrc522, MFRC522::StatusCode status) {
                                                                  ^
~/tonuino/TonUINO-TNG/src/chip_card.cpp:51:66: note: deduced return type only available with -std=c++14 or -std=gnu++14
~/tonuino/TonUINO-TNG/src/chip_card.cpp:58:65: error: 'printPiccType' function uses 'auto' type specifier without trailing return type
 auto printPiccType(MFRC522& mfrc522, MFRC522::PICC_Type piccType) {
                                                                 ^
~/tonuino/TonUINO-TNG/src/chip_card.cpp:58:65: note: deduced return type only available with -std=c++14 or -std=gnu++14
```

Platform local compiler flags should prevent this message
```
$ cat platform.local.txt 
compiler.cpp.extra_flags=-std=gnu++17
```

Please advise.

boerge1, 2024-12-24:
Didn't you copy the file platform.local.txt to the right directory according to the Readme?

ekacoei, 2024-12-24:
My bad. Thank you very much!

closed 2024-12-24 by ekacoei