tonuino/TonUINO-TNG #268: Version 3.3.0: Compile Error when using Rotary-Encoder and Classic HW

10hfh01, 2025-07-19:
## Description: 
Compiling Version 3.3.0 fails with errors, when option for Rotary_Encoder is enabled with TonUINO_Classic.

## Compilation Messages (Path Prefixes removed)
```
In file included from ...\Tonuino\TonUINO-TNG\src\tonuino.hpp:8:0,
                 from ...\Tonuino\TonUINO-TNG\TonUINO-TNG.ino:1:
...\Tonuino\TonUINO-TNG\src\rotary_encoder.hpp:25:15: error: variable or field 'IRAM_ATTR' declared void
   static void IRAM_ATTR changed();
               ^~~~~~~~~
...\Tonuino\TonUINO-TNG\src\rotary_encoder.hpp:25:15: error: expected ';' at end of member declaration
exit status 1

Compilation error: variable or field 'IRAM_ATTR' declared void
```

## My Setup:
Arduino: 2.3.6
Tonuino: 3.3.0
Libs: As recommended in Readme

## Steps to Reproduce:
* checkout Version 3.3.0
* uncomment line 75 in constants.hpp to select the classic hardware setup
* uncomment line 145 to enable rotary encoder

I could fix the compile errors with removing IRAM_ATTR in line 25 of rotary_encode.hpp and line 22 of rotary_encoder.cpp. However, it seems to be a special definition for esp32? Probably needs some #define magic.

Anyway, thanks for implementing the esp32!

(German is fine as well, I wasn't shure what is the preferred language for this project)

boerge1, 2025-07-20:
pull request #269 created

10hfh01, 2025-07-20:
@boerge1 yes the fix works for me (couldn't test esp32 with rotary, they are still on delivery but classic works now). Thanks for the quick fix!

closed 2025-07-22 by boerge1